PR2/pr2_power_drivers #69: Ocean-server battery driver crashes

shahvi, 2018-04-13:
Hi,
I installed the version from the ROS repo using apt install ros-kinetic-ocean-battery-driver
When I try to run the rosnode using rosrun ocean_battery_driver ocean_server, I get an error saying:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range
```
Any thoughts on what might be going on?

Cheers,
Vik

v4hn, 2018-04-14:
I did not expect to see this error for the next 20 years yet..
I assume you synchronized your clocks before?

My first hypothesis would be that your battery controller reports
overly long runtime left. This value is parsed into a rostime without
further checks:
https://github.com/PR2/pr2_power_drivers/blob/kinetic-devel/ocean_battery_driver/src/ocean_battery_driver/ocean.cpp#L666

Patches welcome.

Otherwise just look through all the other places where rostime is used and you will probably find
the source of your error soon..

Did you install ros kinetic on an actual PR2?
If yes, please publish your progress somewhere (discourse.ros.org/the pr2-users ml).
This is very interesting for other groups!

shahvi, 2018-04-14:
If only I had seen the message sooner... I have spent the last few hours writing up a python version of the driver.
I am not installing on a PR2, we are using Ocean Server batteries on an AUV.  The system is currently reporting runtimes of 9999 so that could certainly be the source of the problem.
Thanks for your response.