PR2/pr2_power_drivers #72: ready for release in noetic

davefeilseifer, 2021-03-01:
passes prerelease testing for focal/noetic:

mkdir -p /tmp/prerelease_job
cd /tmp/prerelease_job
generate_prerelease_script.py \
  https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml \
  noetic default ubuntu focal amd64 \
  --custom-repo \
    pr2_power_drivers__custom-2:git:https://github.com/PR2/pr2_power_drivers.git:kinetic-devel \
    rqt_pr2_dashboard__custom-3:git:https://github.com/PR2/rqt_pr2_dashboard.git:kinetic-devel \
  --level 0 \
  --output-dir ./

davefeilseifer, 2021-03-13:
this missed the last sync, can anything be done to push this along @k-okada ?

k-okada, 2021-03-15:
sorry, I am waiting for CI status check,
finally I released the new release into noetic
https://github.com/ros/rosdistro/pull/28644
--
◉ Kei Okada

davefeilseifer, 2021-05-07:
follow-up, it seems like merging PR: https://github.com/PR2/pr2_power_drivers/pull/75 will fix the build issues that are happening.

davefeilseifer, 2021-05-11:
huh, I'm still having trouble with boost dependencies and the Jenkins build. Any ideas @k-okada ?

k-okada, 2021-05-12:
ok, released new version including https://github.com/PR2/pr2_power_drivers/commit/5d0508bd499797f48c4a66ba24732571e79e9b0d
let's see if it works -> 
https://github.com/ros/rosdistro/pull/29498
https://build.ros.org/job/Nbin_uF64__pr2_power_board__ubuntu_focal_amd64__binary/58/consoleFull

BTW, I think https://github.com/PR2/pr2_power_drivers/commit/20140e96069c0e66d2d55fd4de9bad0d0bb12dd9 is not required, and I am afraid there PR2 whcih still uses indigo.

v4hn, 2021-05-12:
> BTW, I think 20140e9 is not required, and I am afraid there PR2 whcih still uses indigo.

We bumped the cmake version throughout MoveIt a while ago as well: https://github.com/ros-planning/geometric_shapes/pull/129
The `3.0.2` is kinetic-compatible.

k-okada, 2021-05-20:
released new version including https://github.com/PR2/pr2_power_drivers/commit/8e475eca88863749dbf46663721f161732b60446 to  https://github.com/ros/rosdistro/pull/29582

@davefeilseifer I prefer to update code via PullRequest so that we can receive notification

davefeilseifer, 2021-05-21:
fair enough, will remember for next time